IlliniHyperloopComputing/Pod #101: Develop network data protocol -- What do we need to send, how do we send it?

rgw3d, 2018-12-06:
We need to send data to and from our vehicle. But we need to answer a couple of questions first:

- What do we need to send from the Pod? ALL data, or just aggregate data? Should this be configurable?
- What format do we send from the Pod? packed as tightly as possible in a few number of bytes? JSON?
- How do we send commands to the Pod? What format? 
- Should we always use TCP? Is UDP useful for some emergency commands?

rgw3d, 2019-01-16:
I'm in favor of a minimal, non-JSON format. I can be convinced otherwise. However, it is somewhat tricky to deal with JSON on the main controller side of things, while dealing with structs on the webserver side are not as difficult.

rgw3d, 2019-01-16:
I think most data should be sent via TCP. Minimal effort. I don't see an immediate need to try and develop our own protocol

rgw3d, 2019-01-16:
Data and its type:
Overall I want to stay away from floats. I know we have enough processing power where it shouldn't matter, but I'd rather stick to ints

Pod State in the state diagram: [uint8], less than 256 states

Pod output GPIO [uint8] bit field of all GPIO outputs and if they are on or not

Overall Kinematics used in decision making:
Overall Position: [int32], measured in millimeters [-500, 1300000]. Track is 1250 km long
Overall Velocity: [int32], measured in millimeters per second  [-200000, 200000]. 200m/s maximum
Overall Acceleration:  [int32], measured in millimeters/second/second, [-200000, 200000]. 200m/s/s ~ 20g

Pressures of all pneumatics (x5 pneumatic lines, x1 battery box), Raw ADC values [uint16]

Orange Tape Sensor counts (x2, both left and right), [uint8] less than 256 tapes
Orange Tape time deltas: (x2, both left and right), [uint32] (whatever time is measured in)

Optical Sensor counts (x2, both sensors), [uint32] lots of rotations
Optical Sensor time deltas (x2, both sensors), [uint32] (whatever time is measured in)

Temperature from 5 points on vehicle: [uint16], Raw ADC values

Accelerometer raw data, 3 points [uint16], raw ADC values

**Battery Data**
Cell voltages from _all_ cells (30 points), [uint16 or 32] depending on data type 
Temperature from all points measured on battery (40 points) [uint16]
Current [int32] exact units depend on what the sensor outputs
State of Charge (SOC) percent from BMS [uint8]
Operation bits [uint8] bit field of [+ relay, - relay, precharge relay... anything else we need]

**Motor Controller Data**
Motor temperature [uint16]
Motor RPM [uint32]
Motor Position (cumulative) [uint32] if available 
Motor measured current [uint32]
Motor measured DC bus voltage [uint32]
Other interesting motor stats

rgw3d, 2019-01-16:
Overall the data format could look something like this:

[uint8 ID] | [data]

Where the ID indicates how much data and of what type it will be. 
If we have 40 different IDs, then we will need 40 different statements on the webserver to parse the data

Is it a little tedious? yeah. But we have used it in the past, and it is about as compact as you can make it

the strategy for reading data:
Read one byte
That byte indicates what data is incoming, and we can lookup the size of that data
Read the rest of the data based on the given size

rgw3d, 2019-01-16:
Commands for the Pod:

**General Commands**
Transition to state X
Emergency Stop 

**Functional test specific**
Enable/ Disable HV bus
Enable/ Disable Brakes
Control motor speed/torque

**Command Packet Format:**
[uint8 ID] | [command data uint8]
 
For example, say the Transition command had ID 3, and we wanted to go to state 4: [3] | [4]
Or we wanted to set the motor torque manually to 7, and it had ID 2:  [2] | [7] 
Something like that

**UDP emergency**
Instead of TCP, we can use repeated UDP packets to quickly signal an emergency
Send a uint8 message with `0xAA=0b10101010`. Something that is easily identified (not 0x00 or 0xff). We can send many of these messages

schmidtwmark, 2019-01-16:
Note that we already have a network protocol for sending commands to the Pod, but this protocol is (frankly) not very good and your suggestions are much better. Note that because it already exists, changes to the Network Command stuff will likely necessitate changes to the simulator/possibly the simulator test cases.

rgw3d, 2019-03-30:
Important safety things to show:

Battery healthy (no isolation faults, good voltage)
Temperatures good (battery, pod, motor, motor controller)
Pressure good
State we are in
What outputs we have on (which relays are on, is highvoltage exposed, pneumatic valve state)
All sensor output is good

PR #159 (2019-08-09): New Frontend
Here is the new and improved BaseStation. There is still some work to be done but it is looking and acting a lot better. 
ToDos:
* UDP server needs work
* Frontend does not detect if connection is lost
* State needs to be shown more prominently
* More stats need to be shown
* Stat danger calculations for color changes
* Finish command code implementation
* Testing on heavy load
* Beautify

This will resolve #110 . It solves the I2c issues
This will close #101
closed 2019-08-09 by rgw3d