IlliniHyperloopComputing/Pod #104: ATtiny-85 Watchdog Development

rgw3d, 2019-01-15:
Development of the watchdog for the main controller will be done on an ATtiny-85 using Arduino Studio.
https://www.microchip.com/wwwproducts/en/ATtiny85

**The requirement from SpaceX**: to be fail safe (shut off motor and activate brakes) in case of power failures, software failures, communication, or sensor failures.

**The problem that this device solves:** if the main controller (the BBB) actively detects a sensor fault or communication fault it will go into a fail safe state. If the power goes out then the electronics are already setup to go into a fail safe state. _However,_ if our software faults out, or the kernel crashes, or something bad happens then we need something outside of the Main Controller to catch it.

**The Solution via a heartbeat:** To see if the main controller is functioning, we can have the main controller toggle an output signal in it's main loop. If software crashes, this toggling stops. Then the watchdog simply needs to watch to see the heartbeat and verify the main controller is working in a bounded sample time (like 50 or 100 milliseconds)

**What the heartbeat looks like:** The heartbeat looks like a 3.3v GPIO toggling between on and off. What matters is detecting that the main controller has changed its output from "high" to "low" and vice versa. As long as a change has been detected within time frame X, then we are okay. We will need to identify what the time frame X (20 milliseconds, 50 milliseconds, 100 ??) in order to quickly identify a crashed main controller.

Overall code shouldn't be longer than 1 page/screen. The somewhat tricky part is detecting a heartbeat

Roughly the pseudo code will look like:
```
Turn on

turn watchdog output "healthy" LOW

wait for X amount of signal changes (heart beats) within a short time period (to make sure power cycles or other transients doesn't accidentally turn it on)

turn watchdog output signal HIGH

wait until heartbeat fails

turn watchdog output "healthy" LOW

while(1);
```

AFederici, 2019-02-06:
In regards to the final line, do you really mean to do nothing after a failure has been spotted or do you mean to continuously output an "Unhealthy" signal to make sure that all of the other components know something is wrong?

AFederici, 2019-02-06:
while(1){
output unhealthy
} 

or

output unhealthy
while(1){
nothing
}

rgw3d, 2019-02-08:
All you need to do is to set the output LOW, and then `while(1){ }`. You don't need to keep setting the output, it retains its value.

closed 2019-02-22 by rgw3d