IlliniHyperloopComputing/Pod #107: CAN Bus Application layer

rgw3d, 2019-01-16:
Development of the CAN bus such that it is easily usable without the command line utilities `candump` and `canwrite`. Needs to be a C/C++ solution, that get CAN frames and make it available to the rest of the system. It also needs to be able to write to CAN as well. 

There will be the Motor Controller, BMS, and maybe an additional temperature sensor that connects over CAN and feeds data to the BMS.

Specifics as to what will need to be parsed will come later, but at a minimum, a single CAN frame needs to but put into a data structure of some kind along with its ID. Then different functions can get the frame and translate it to meaningful data.

The same thing applies to sending data: at a minimum, a CAN frame must be able to be specified and passed into a function that sends it out. More wrappers can be built up around this function to make it easy for the outside to specify a "motor increase speed" command or something like that.

schmidtwmark, 2019-01-16:
The functionality for polling from the CAN bus should occur within CANManager's refresh() function. CANManager can be expanded upon to handle sending commands as well if necessary, but keeping it as a SourceManagerBase object will save a lot of trouble with making commands and handling data elsewhere in the codebase.

rgw3d, 2019-02-14:
Might want to use this as a guide: https://github.com/linux-can/can-utils/blob/master/candump.c

closed 2019-03-15 by rgw3d