IlliniHyperloopComputing/Pod #150: Full Test List

rgw3d, 2019-04-26:
We have a lot of unit tests right now, which is great

But we need to document them better, and SpaceX is looking for that.

What I mean is we ought to have a document that lists tests like this:
Test Group 1: Verification of State Machine
Test 1.1 Verify Transitions into SafeMode: Located in StateTests.cpp
Test 1.2 Verify Transitions into Functional Test: Loacated in StateTests.cpp
Test 1.3......

anshulshah957, 2019-04-30:
Would this be in a README inside the tests directory, or just like a separate word doc?

rgw3d, 2019-05-02:
Likely the README or something similar. It is much closer to the code this way, and there isn't ever an issue like "where is that file?"